Clank50AE/Clanks-Klipper-Configs #4: Neopixel Sidewinder x1 config

akshad09, 2021-04-17:
I love your printer.cfg for the sidewinder x1. Th main issue is that I'm trying to make the led neopixel work on the extruder but I don't know the pin name on how to configure it. Can anyone help?
Here is the config file just in case:

[stepper_x]
step_pin: ar54
dir_pin: ar55
enable_pin: !ar38
microsteps: 16
rotation_distance: 40
endstop_pin: ^!ar3
position_endstop: 0
position_max: 310
homing_speed: 40.0

[stepper_y]
step_pin: ar60
dir_pin: ar61
enable_pin: !ar56
microsteps: 16
rotation_distance: 40
endstop_pin: ^!ar14
position_endstop: 0
position_max: 310
homing_speed: 40
homing_positive_dir: false

[stepper_z]
step_pin: ar46
dir_pin: !ar48
enable_pin: !ar62
microsteps: 16
rotation_distance: 8
endstop_pin: ^!ar18
position_endstop: 0
position_min: -3
position_max: 400
homing_speed: 30
second_homing_speed: 1

[stepper_z1]
step_pin: ar36
dir_pin: !ar34
enable_pin: !ar30
microsteps: 16
rotation_distance: 8

[extruder]
step_pin: ar26
dir_pin: ar28
enable_pin: !ar24
microsteps: 16
rotation_distance: 7
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: ar10
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog13
min_extrude_temp: 190
min_temp: 5
max_temp: 280
max_extrude_only_distance: 500
max_extrude_cross_section: 2.0

[heater_bed]
heater_pin: ar8
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog14
min_temp: 5
max_temp: 150

[fan]
pin: ar9
kick_start_time: 0.500

[heater_fan my_nozzle_fan]
pin:ar7
heater: extruder
heater_temp: 50.0
fan_speed: .7

[mcu]
#make sure this reflects your serial port you found when flashing klipper
serial: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
pin_map: arduino

[printer]
kinematics: cartesian
max_velocity: 200
max_accel: 3000
max_z_velocity: 50
max_z_accel: 100
square_corner_velocity: 5 

[bed_screws]
screw1: 55,55
screw1_name: front left
screw2: 255,55
screw2_name: front right
screw3: 255,255
screw3_name: back right
screw4: 55,255
screw4_name: back left
speed: 100.0

[screws_tilt_adjust]
screw1: 22,83
screw1_name: front left
screw2: 222,83
screw2_name: front right
screw3: 22,283
screw3_name: back left
screw4: 222,283
screw4_name: back right
speed: 100.0
screw_thread: CW-M5		
			 
[gcode_macro START_PRINT]
# Reference https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Reference.md#gcode_macroA
# On how to override default parameters
default_parameter_BED_TEMP: 60
default_parameter_EXTRUDER_TEMP: 200	
								 	
gcode:
    # Home the printer
	G28 				 		  						 
    # Use absolute coordinates
    G90
    # Move the nozzle near the bed
    G1 X0 Y0 Z5 F3000
    # Move the nozzle very close to the bed
    G1 Z0.15 F300
    G92 E0 ;Reset Extruder
    G1 X20 Y5 Z0.3 F5000.0			   
    G1 Z0.3 F1000 				    
    G1 X200 Y5 F1500.0 E15 			   
    G1 X200 Y5.3 Z0.3 F5000.0			    
    G1 X5.3  Y5.3 Z0.3 F1500.0 E30 			
    G1 Z3 F3000 				     
    G21 ; set units to millimeters
    G90 ; use absolute coordinates
    M83 ; use relative distances for extrusion
    G92 E0

[gcode_macro END_PRINT]
gcode:
    # Turn off bed, extruder, and fan
    M140 S0
    M104 S0
    M106 S0
    # Move nozzle away from print while retracting
    G91
    G1 X-2 Y-2 E-3 F300
    # Raise nozzle by 10mm
    G1 Z30 F3000
    G90
    # Disable steppers
    M84

[pause_resume]
recover_velocity: 50.00

[virtual_sdcard]
path: ~/gcode_files

[display_status]

[gcode_macro CANCEL_PRINT]
rename_existing: BASE_CANCEL_PRINT
gcode:
    TURN_OFF_HEATERS
    CLEAR_PAUSE
    SDCARD_RESET_FILE
    BASE_CANCEL_PRINT

[gcode_macro FILAMENT_LOAD]
gcode:
  # store min extrution temp in variable
  {% set minTemp = printer.configfile.config["extruder"]["min_extrude_temp"] %}
  {% if printer.idle_timeout.state != "Printing" or printer.pause_resume.is_paused|lower == "true" %}
    SAVE_GCODE_STATE NAME=STATE_LOAD_FILAMENT
    ## Move to waste bin
    G28                 ; home if not already homed
    G90                   ; absolute positioning
    #G0 X55 Y352 Z30 F9000 ; park nozzle at brush bin
    # check the Extruder temp is above minimum extrution temp
    {% set extruder_target = printer.extruder.target %}
    _LCD_KNOB COLOR=BLUE
    {% if printer.extruder.temperature|int < minTemp|int %}
      {action_respond_info("Extruder Temp to low heat to %2dC" % (minTemp|int + 30))}
      M109 S{minTemp|int + 30} ; heat extruder and wait
    {% endif %}
    _LCD_KNOB COLOR=RESTORE
    M83                   ; set extruder to relative
    G1 E90 F1800          ; quickly load 90mm filament
    G1 E50 F300           ; slower extrusion for hotend path
    G1 E-2 F1500          ; retract 
    SAVE_VARIABLE VARIABLE=filament_loaded VALUE='"true"'
    # clean nozzle
    _WIPE
    G0 Z30 F1800
    G0 X55 Y352 F9000 ; park nozzle at brush bin
    # restore old extruder temperature
    M109 S{extruder_target}
    _PRINT_AR T="RUNOUT Motion Sensor Enable: true"
    SET_FILAMENT_SENSOR SENSOR=runout ENABLE=1
    _PRINT_AR T="Filament loaded"
    RESTORE_GCODE_STATE NAME=STATE_LOAD_FILAMENT
  {% else %}
    _PRINT_AR T="Filament loading disabled while printing!"
  {% endif %}

[gcode_macro FILAMENT_UNLOAD]
default_parameter_DISTANCE: 75
gcode:
  # store min extrution temp in variable
  {% set minTemp = printer.configfile.config["extruder"]["min_extrude_temp"] %}
  {% if printer.idle_timeout.state != "Printing" or printer.pause_resume.is_paused|lower == "true" %}
    SAVE_GCODE_STATE NAME=STATE_UNLOAD_FILAMENT
    _PRINT_AR T="RUNOUT Motion Sensor Enable: false"
    SET_FILAMENT_SENSOR SENSOR=my_sensor ENABLE=1
    _LCD_KNOB COLOR=BLUE
    # check the Extruder temp is above minimum extrution temp
    {% set extruder_target = printer.extruder.target %}
    {% if printer.extruder.temperature|int < minTemp|int %}
      {action_respond_info("Extruder Temp to low heat to %2dC" % (minTemp|int + 30))}
      M109 S{minTemp|int + 30} ; heat extruder and wait
    {% endif %}
    # Ball up the filament tip and retract out past the extruder gears
    _LCD_KNOB COLOR=RESTORE
    _FILAMENT_BALL WAIT=3
    M83 ; Relative extrusion
    G1 E{DISTANCE|float * -1} F3000
    M400
    SAVE_VARIABLE VARIABLE=filament_loaded VALUE='"false"'
    # restore old extruder temperature
    M109 S{extruder_target}
    _PRINT_AR T="Filament unloaded"
    RESTORE_GCODE_STATE NAME=STATE_UNLOAD_FILAMENT

  {% else %}
    _PRINT_AR T="Filament unloading disabled while printing!"
  {% endif %}
  
[filament_switch_sensor my_sensor]
#Move filament runout sensor from TFT to X+ endstop pin. Umcomment all but this line if you want to use this
pause_on_runout: True
runout_gcode: PAUSE
insert_gcode: RESUME
switch_pin: ar2

**### [neopixelmy_neopixel]
pin: EXP1_6
chain_count: 3
color_order: GRB
initial_RED: 0.4
initial_GREEN: 0.4
initial_BLUE: 0.4**

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [extruder]
#*# control = pid
#*# pid_kp = 21.465
#*# pid_ki = 0.980
#*# pid_kd = 117.521
#*#
#*# [heater_bed]
#*# control = pid
#*# pid_kp = 48.362
#*# pid_ki = 0.638
#*# pid_kd = 915.854

Dragi2k, 2021-05-14:
The LED is not a Neopixel.
Neopixel = + , Data , -
LED on Sidewinder = B+, G+ ,R+ , -
The PINS should be the Servo 2 Pins

Dragi2k, 2021-05-15:
https://github.com/KevinOConnor/klipper/issues/3376#issuecomment-702564677

her is your code

closed 2021-07-14 by Clank50AE

edoluck90, 2023-08-02:
Hello people someone has managed to make it work ? i mean the stock led mounted on x1 ?